AguaClara/Fluoride-Auto #1: Task List/Map

djs469, 2018-02-05:
[Task List 2018.docx](https://github.com/AguaClara/fluoride/files/1697130/Task.List.2018.docx)

Task List 
-	Set up a one reactor system and make sure probes and other measuring equipment are usable and calibrated- Everyone (2/9/18)
-	Run experiments comparing the results with the tap water standards to verify last semester’s results- Tigran (2/23/18)
-	Set up concentrated stock solutions for the experiments so that the flowrate of fluoride used during the experiment is lower, meaning less time spent changing the stock tanks- Desiree (2/14/18)
-	Run experiments varying the upflow velocity through the sedimentation tube and analyze results to measure removal efficiency and obtain an optimal upflow velocity- Victoria (3/6/18)
-	Set up a two reactor countercurrent system- Philip (3/25/18)
-	Run experiments using the countercurrent system and measure differences in removal efficiency and other parameters for reactor performance from the one reactor system- Victoria (4/24/18)
-	If time allows, design and begin fabricating pilot scale reactor that is electricity free (0.1 L/s)- Everyone (5/9/18)

auggiekl, 2018-02-11:
Make  each task its own issue

nlm42, 2018-02-21:
Please create this task list in the "project" page and call these tasks as individual issues. [Here is an example](https://github.com/AguaClara/1-LPS/projects/1).

closed 2018-02-23 by DesireeJSausele