AguaClara/Fluoride-Auto #13: Length Flocculator?

tm545, 2018-02-27:
Could you confirm our length of flocculator, because on your flocculator values drive page shows two separate values for this length, one being 13.709m for 1mm/s and the other as 11.827m for the 3mm/s flocculator. Our flocculator was a 1.5mm/s so we're not sure which value is accurate or if the fluoride team had a flocculator with a length between these two values.
Thanks in advance.

TanviNaidu, 2018-02-27:
You will need a different flocculator length, in between those two values. There is a place in the mathcad file to input sed tank upflow velocity (in your case, 1.5 mm/s). If you want help with mathcad while transitioning to python, let me know.

TanviNaidu, 2018-02-28:
Hey team, 
3:30 on Friday doesn't work for me, but I am free in the morning on Friday till 11:15. 
What are your other meeting times?

vxz3, 2018-02-28:
Monday, Tuesday, Thursday 4:30-6:30 & Wednesday 2-4

vxz3, 2018-03-05:
Do you have any time to meet this week?

TanviNaidu, 2018-03-05:
Yes, I can meet you on Wednesday at 2!

vxz3, 2018-03-05:
Great! Thanks!

pakpan, 2018-03-07:
Hi Tanvi,

Were you still planning on coming in to the lab to meet with us today?

closed 2018-04-17 by pakpan