AguaClara/Fluoride-Auto #14: Issue Conforming to Report Guidelines

pakpan, 2018-03-06:
Although we are considered a research team, we have been unable to do meaningful experimental research this semester due to probe issues. As a result, we have shifted our semester goals to be more geared towards the design of a gravity-powered Fluoride removal system. Because of this, we no longer fit into the general research report format. In addition, we are likely not confident enough or learned enough in fusion 360 to provide drawings for our design as a fabrication team would, nor did we plan in advance to have such things done by this deadline. What do you suggest that we should do for this Friday deadline?

nataliemottl, 2018-03-06:
Hello!

So let's start by pushing the due date back to Sunday night at 11:59pm. That hopefully will alleviate some pressure with the due date. If absolutely necessary, we can discuss a further extension, but try to look ahead better in the future.

We have a couple options with grading:
1. Extensive manual with design considerations, code, etc + hand-drawn drawings/plans (or try fusion)
2. Extensive manual with a proposal for your design (2-3 pages) - this could be a formal explanation of your design in a persuasive sense or an organizational sense
3. Extensive manual + literature review/previous work plus design considerations
(All increase value of manual and sub in a different thing for the report)

I am not familiar enough with your exact focus/work to properly advise which one of these to choose. Please talk to your team ASAP and get back to me tonight on which option works better or counter propose a different option. Then I will update your rubrics and get you guys rocking and rolling!

pakpan, 2018-03-06:
Would you want everything to be included under a "Manual" section, or should we do something similar to the report format but instead of a procedure and methods section we provide a section for design considerations instead?

nataliemottl, 2018-03-06:
Which one makes more sense to you?

pakpan, 2018-03-06:
Not sure yet, I think that for now we'll plan to do a report format that will contain the literature review and previous work since we already had those sections, and then include a manual section containing things like our design aspects and the code we've written thus far along with our drawings and plans.

nataliemottl, 2018-03-06:
That sounds good, I'll change your rubric tonight to reflect this. Keep me updated!

pakpan, 2018-03-06:
Thanks Natalie!

nataliemottl, 2018-03-08:
See [updated rubric here](https://docs.google.com/document/d/1OBtgsZYs4AIPQiuJ3-eQooiBD14wC8a_8FyEmbjIxGQ/edit)

DesireeJSausele, 2018-04-17:
We still haven't done much actual research due to probe issues. Instead, we've been focusing on fabrication of our gravity powered design.  The rubric for Report 2 follows the general guidelines for a research team, but we were wondering if we could change it to be similar to the rubric for Report 1.

nataliemottl, 2018-04-18:
I have updated the rubric for report 2. Similar to report 1 but now with a section for fabrication considerations.

closed 2018-05-11 by vxz3