AguaClara/Fluoride-Auto #17: Construct gravity powered fluoride removal system

DesireeJSausele, 2018-03-16:
First confirm that calculations of dimensions are correct and then create a to scale schematic. Collect the necessary materials and begin construction (may need to consult with Paul)

*note: if it is not feasible to build the system at our desk, then we will need to find an alternate location.

vxz3, 2018-04-19:
Began constructing gravity powered reactor based off Fusion design. Built base this week and starting making the shelves by cutting 80/20 bars and PVC sheets. Slow and steady progress.

vxz3, 2018-04-26:
Finished construction of base/support structures and beginning to attach nalgene bottles, buckets, flocculators, etc.

vxz3, 2018-05-09:
Attached tubing and checked for leaks in the system, replacing the connections that were faulty.

vxz3, 2018-05-09:
Tested the reactor and water does flow in the reactor and out through the effluent line.

vxz3, 2018-05-11:
Replaced some connections that were leaking.

vxz3, 2018-05-11:
Now that the gravity powered reactor is up and running, the summer team can start running tests!

closed 2018-05-11 by vxz3