AguaClara/Fluoride-Auto #26: Microbore tubing

vxz3, 2018-04-27:
Hey Casey,

I just wanted to let you know that we just ordered some microbore tubing with part number: EW-06417-21.

pakpan, 2018-04-27:
To clarify, we mean that we put the order into the Fittings List order excel sheet

caseyching13, 2018-04-28:
Thank you for notifying me and for the clarification! Just a heads up, it took a week for the last order of microtubing to arrive (the order took a few days to be approved by Cornell). There is microtubing that just arrived on Thursday with the sizes 0.042"ID x 0.066"OD and 0.052"ID x 0.076"OD. Nevertheless, I will place the order for the microtubing as soon as possible (eshop is currently down for maintenance)!

vxz3, 2018-05-09:
5/2/18: Microbore tubing was too small to use current 1/4'' caps so Paul used the smallest drill bit he had to drill us a hole for the current microbore tubing.

closed 2018-05-11 by vxz3