AguaClara/Fluoride-Auto #35: Fabrication for new reactor

ChingPangggg, 2018-06-04:
Started fabrication of new reactor by cutting out 2 sets of PVC pipes for the tube settler (90 cm) and floc weir (40 cm) according to 2017 Fall HRS team final report.
Drill bit: 1/2 in
Thread: 1/4 in

cheertsang, 2018-06-06:
June 5, 2018
Created bend in reactor using welder (60 degrees from horizontal)
Drilled hole into reactor for floc weir using circle saw and drill press at 55 degree angle. 
Started welding floc weir pipe to tube settler.

cheertsang, 2018-06-07:
June 6, 2018
Finished fabricating sed tube. Checked for water-tightness.

cheertsang, 2018-06-07:
June 7, 2018
Assembled gravity-powered system.
- Added sed tube

ChingPangggg, 2018-06-08:
June 7, 2017
Added valves to the exits of the constant head tanks for easier refills

cheertsang, 2018-06-08:
June 8, 2018
Finished welding sed tubes. 
Leaky bottom on sed tube on gravity powered setup - fix by screwing bottom tighter, adding more Teflon tape.

ChingPangggg, 2018-06-13:
June 11, 2018
Fixed leak on sed tube of the gravity powered setup. Water tested the two new reactors and no leak was found.

closed 2018-06-13 by ChingPangggg