AguaClara/Fluoride-Auto #36: Set up bench system

cheertsang, 2018-06-07:
Set up fluoride bench system, compare flow rate out of sed tube to gravity powered setup.
- Determine fluoride stock solution
- Determine PACl stock solution
- Determine pump speeds for fluoride and PACl 
- Calculate necessary water pump speed for 1.5 mm/s upflow velocity in sed tube
- Write Python code for all of the above

cheertsang, 2018-06-08:
**Determine fluoride stock**
Literature shows highest reported levels of fluoride in groundwater are up to 21 mg/L. 
- start with 5 mg/L
- 10 mg/L
- 15 mg/L
- 20 mg/L
Target range of fluoride is: 0.6 to 1.5 mg/L (should reduce fluoride concentration to this amount)
[(Karthikeyan and Lakshmanan, 2011)](https://github.com/AguaClara/fluoride/files/2082254/Fluoride_book_chapter.pdf)
Fluoride solution: 10,000 ppm 
- **2.5 mL fluoride solution per 1 L of water** - yields **24.876 mg/L** fluoride stock
- **1.5 mL fluoride solution per 1 L of water** - yields **14.925 mg/L** fluoride stock
- **1.0 mL fluoride solution per 1 L of water** - yields **9.95 mg/L** fluoride stock
- **0.5 mL fluoride solution per 1 L of water** - yields **4.975 mg/L** fluoride stock

cheertsang, 2018-06-08:
June 8, 2018
Started calibrating fluoride probe. 
Measured fluoride probe voltage with solution of 50 mL DI water, 50 mL TISAB solution, and 1 mL 1000 (?) mg/L fluoride solution. E1 = 0.06657, E2 = 0.009. Slope = -0.05757. Slope was within the range of -0.056 V, indicating probe was working correctly. 

[Table for calibration](https://docs.google.com/spreadsheets/d/1Qdzn8rtu0ubeyHFeoxpHNJPTBkcbFvO21DLpOUB-Flc/edit?usp=sharing).

cheertsang, 2018-06-12:
June 12, 2018
Need to calculate pump speeds for fluoride concentrations we want through the system. 
Fluoride pump uses orange-yellow tubing size, [conversion](https://confluence.cornell.edu/display/AGUACLARA/Auto+Tutorial+for+Peristaltic+Pumps) = 0.019 mL/rev.

cheertsang, 2018-06-13:
June 12, 2018
**Experiment setup**
Vary fluoride stock concentration, keep pump speed constant. 
Fluoride pump speed = 20 rpm
Fluoride stock: 
- 10,000 ppm solution
- **4 mL F- solution per 1 L water**
- yields fluoride stock concentration of **39.84 mg/L**

Qsys*Csys = Qstock*Cstock
- Qsys = 1.013 mL/s
- Qstock = 0.000049667 L/s
- Cstock = 39.84 mg/L
- Concentration of fluoride in system = **19.53 mg/L**

Repeat calculations for 3, 2, 1 mL F- solution per 1 L water

ChingPangggg, 2018-06-13:
[image]
Calibration of Fluoride pump

cheertsang, 2018-06-20:
June 20, 2018

Measured actual flow rates through system.
Coagulant pump: 2 mL/ 5 min 58 s = 0.005586592179 mL/s
- Set points on ProCoDA: Flow rate of coagulant = 0.0009719 mL/s, 0.019 mL/s
- Pump running at 14 rpm
- Actual pump speed on ProCoDA = 0.031 mL/s

Verified actual flow rate by using graduated cylinder and timing flow rate per volume:
Ran coagulant pump manually at 6 rpm
- 1.9 mL/ 12 min 53.51 s = 0.002456335406 mL/s
- Target flow rate = 0.0019 mL/s
Ran coagulant pump manually at 40 rpm
- 4 mL/4 min 0.15 s = 0.01665625651 mL/s
- Target flow rate = 0.01267 mL/s

cheertsang, 2018-06-20:
June 20, 2018

Changed out 6-600 rpm pump to 1.6-100 rpm pump for coagulant for better accuracy at low flow rates. 

- Measured flow rate = 1 mL/9 min 55 s = 0.001680672269 mL/s
- Target flow rate = 0.001583 mL/s (5 rpm)

Verified water pump speed. 
- Measured flow rate = 9.6 mL/10.3 s = 0.932038835 mL/s
- 9 mL/10.37 s = 0.8678881389 mL/s
- Target flow rate = 0.7601 mL/s (57 rpm)

Changed water pump speed set point: 0.6601 mL/s
- Measured flow rate = 8.2 mL/11.64 s = 0.704467354 mL/s

Changed water pump speed set point: 0.7001 mL/s
- Measured flow rate = 9.6 mL/11.61 s = 0.826873385

Changed water pump speed set point: 0.6801 mL/s
- Measured flow rate = 9 mL/11.85 s = **0.7594936709 mL/s**

cheertsang, 2018-06-22:
Verified flow rates through water pump, fluoride pump, and coagulant pump.
Adjusted flow through water pump as necessary to obtain upflow velocity of 1.5 mm/s.

closed 2018-06-22 by cheertsang