AguaClara/Fluoride-Auto #37: Changing pump speeds on ProCoDA

cheertsang, 2018-06-12:
Hi @WPennock @jillianwhiting ! We were wondering if you could help us with ProCoDA. We are trying to change the pump speeds, but the pumps are not responding to the changes we make in ProCoDA. A couple things we tried:
- Changed the set points for pump speed.
- Checked that pumps are running on external input
- Checked that pump ID was correct
- Checked that pump ID and flow rate was selected in water pump set point
- Calibrated pumps
[image]
[image]

Do you have any suggestions on what else we could try?
Thank you!

jillianwhiting, 2018-06-12:
Cheer,
How far off is it? Like only one or two rpm or a lot. If it is very far off try another pump cable. If they are pretty close sometimes when the pump is off, it is still running at like 1 rpm and then if you hit the stop button when you run it via procoda again it will be right. Those are the two problems I know.

Jill

cheertsang, 2018-06-12:
Hi Jillian: 
It is running at the same speed no matter what we change the pump speed set point to. Right now, it is running at 22 rpm. We set the pump speed set point to 1.013 mL/s, and at this rate, we calculated that the pump should be running at 76 rpm. We also tried inputting different pump speed values, but it didn't change the pump speed on the actual pump.

jillianwhiting, 2018-06-12:
Cheer,

That sounds like an issue with either the pump cord or the pump itself. I would try to see what happens when you use a different cord or pump.

Jill

ChingPangggg, 2018-06-13:
Hi Jillian.

We tried to change the pump cords but nothing really happened. Also, all three pumps we tried connected to ProCoDA, but none of them respond to the changes we made in ProCoDA. Should we switch out all three pumps?

cheertsang, 2018-06-13:
@pakpan @tm545 @DesireeJSausele Did you guys have this issue with the pumps last semester?

monroews, 2018-06-13:
Need more details.
DO the pumps turn on?
You are sending them the command to run at full speed. Are they running at full speed?

monroews, 2018-06-13:
When you changed the inputs, what did the target pump speed change to? In the image below the target pump speed is 1. Does the pump run at full speed with this setting?
[image]

cheertsang, 2018-06-13:
We tried putting a pressure sensor, and the ports for the sensors don't work either, so we think the problem is the box for the ports.

monroews, 2018-06-13:
Is your PRoCoDA box plugged in to power?
Does your Fluoride sensor work?
Try a different workstation and see if you can control a pump (WITHOUT CHANGING ANYTHING) at that station (perhaps use 2 stage coag station).

Once you have a working station then swap parts to figure out what isn't working at the fluoride station.

monroews, 2018-06-13:
Check to see if the PRoCoDA box is plugged in to power.

cheertsang, 2018-06-13:
Unplugging and plugging it back in again seemed to solve the issue of the ProCoDA box - now our fluoride sensor is working. The only issue we have now is changing pump speeds...we are going to try using our pumps with the 2 stage station to see if that works.

monroews, 2018-06-13:
The power supply on the ProCoDA box turns off if it is overloaded. THis can happen if someone plugs a valve into a sensor port. I believe you reset the power supply by unplugging the power and plugging the power back in again. I'm assuming it was the power that you unplugged....

ChingPangggg, 2018-06-13:
Pump speeds were fixed by changing set points and outputs in ProCoDA.

closed 2018-06-13 by ChingPangggg