AguaClara/Fluoride-Auto #38: Fluoride Concentration

cheertsang, 2018-06-13:
Hi @monroews !

We are planning to run bench experiments with various system concentrations of fluoride. We wanted to run tests using 20 mg/L, 15, 10, and 5. (This is because the highest reports amounts of fluoride in literature were 20 mg/L, while most groundwater resources had levels around 5 mg/L [(Karthikeyan and Lakshmanan, 2011)](https://github.com/AguaClara/fluoride/files/2082254/Fluoride_book_chapter.pdf). 

However, in order to have a system concentration of 20 mg/L, if we run the fluoride pump at 40 rpm (we want to keep the fluoride pump running at a low flow rate because our calculations assume a negligible flow rate input from the coagulant and fluoride pumps), this would require us to have a stock concentration of fluoride of 218 mg/L **(280 mL of 1000 ppm F- solution per 1 L of water)**. We double-checked our [calculations](https://github.com/AguaClara/fluoride/blob/master/SummerFluorideCalculations.md), but we think this concentration of fluoride seems REALLY high, so we were wondering if our assumption of 20 mg/L for the system concentration was correct. 

Thank you!

monroews, 2018-06-13:
I suggest that you work with @HannahSi since she is also learning how to set up a python file to do these calculations.
The equation that matters is Cstock*Qstock = Cdose*Cdose
Your goal is to design a stock that allows your fluoride stock pump to run as slowly as possible (ideally close to 5 rpm for the LOWEST dose that you want to apply. 

As a point of reference, the coagulant stock we use is 70 GRAMS per liter and this is the concentration that Cornell water treatment plant uses. Their dose is probably around 5 mg/L, so roughly a factor of 1000! We haven't achieved that in the lab and I think it is because student teams have been unwilling to use the smaller diameter pump tubing that we have.

cheertsang, 2018-06-14:
We actually did create a python file [here](https://github.com/AguaClara/fluoride/blob/master/SummerFluorideCalculations.md) to do these calculations. We designed a stock so that our dose concentration is 20 mg/L. However, this stock would require a LOT of fluoride, since it is diluted by the water flow through the system. In addition, if we use the smaller diameter tubing (orange-yellow) and run the pump at 5 rpm, it would take even more fluoride to design a stock to have 20 mg/L, or even 5 mg/L, running through our system. 

We calculated the dose concentration iteratively, by plugging in different pump speeds (pump_speed) and stock concentration (v_stock) to calculate the dose concentration (c_solution). 
```
pump_speed = 5*(u.rpm)
orange_yellow = 0.019*(u.milliliter/u.revolutions)
oy_flowrate = orange_yellow.to(u.liter/u.revolutions)*(pump_speed).to(u.revolutions/u.s)
Q = 1.013 * (u.milliliter/u.second) 

fluoride_stock = 1000 * (u.mg/u.L)
v_stock = (675 * u.mL).to(u.L)
v_water = 1 * u.L
total_v = v_water + v_stock
c_solution = (fluoride_stock * v_stock)/(total_v)

Q_sys = Q.to(u.L/u.s) #water pump speed, calculated above
Q_stock = oy_flowrate
C_stock = c_solution
C_sys = ((Q_stock*C_stock)/(Q_sys)).to(u.mg/u.L)

percent_flow = (oy_flowrate/Q_sys)*100
#more detailed code [here](https://github.com/AguaClara/fluoride/blob/master/SummerFluorideCalculations.md)
```

ChingPangggg, 2018-06-14:
1. Assume the min pump speed of F- is 5 rpm
2. End Goal: Vary pump speed to adjust system concentration of fluoride while keeping the stock conc. of fluoride the same
3. Change upflow velocity to 1.5 mm/s (1 mm/s will cause the flocs to become a gel-like substance and 2 mm/s would cause the flocs to come out at the effluent)
4. Measure actual flow rate of the fluoride pump using a balance, and use the actual flow rate to recalculate the system concentration of fluoride.
5. Measure actual flow rate of the water pump by timing the rate with a graduated cylinder
6. Start with keeping fluoride pump at 5 mg/L or 5 rpm manually (keep fluoride at a constant in each experiment), and then adjust the different levels of coagulant (ProCoDA controls water pump and coagulant pump)
7. Firstly, assume Qstock, Qsystem and Csystem are known, and then solve for Cstock. (Q1C1=Q2C2)

cheertsang, 2018-06-22:
[Calculations](https://github.com/AguaClara/fluoride/blob/master/SummerFluorideCalculations.md) for fluoride and coagulant concentrations.

closed 2018-06-22 by cheertsang