AguaClara/Fluoride-Auto #39: Gravity Powered System Adjustment

ChingPangggg, 2018-06-14:
Tasks:
1. Measure actual flow rate of the gravity powered system by timing water exiting the effluent
2. Check if the flow rate is at 0.76 mL/s
3. Adjust the height if the flow rate does not agree with the upflow velocity of 1.5 mm/s

ChingPangggg, 2018-06-15:
06/15/2018
First test of flow rate (effluent): 0.4762 ml/s
--> (need to adjust the height to reach 0.76 ml/s)
Shortened the tubing between the T-connection and the fluoride tank

ChingPangggg, 2018-06-19:
Added Bars to the system to make sure the filled constant head tank and fluoride stock tank are stable.

cheertsang, 2018-06-22:
Set up gravity powered system, fixed all leaks and structural instabilities. 
Adjusted height of fluoride constant head tank and height of sedimentation tube to achieve upflow velocity of 1.5 mm/s.

closed 2018-06-22 by cheertsang