AguaClara/Fluoride-Auto #45: Fluoride Probe Calibration

ChingPangggg, 2018-06-22:
6/21/2018

[image]
Carried out fluoride probe calibration using dilutions of 1000ppm solution and tried to verify the accuracy of the measurement, but the numbers seem to be very off. We are wondering the calculations from volts to mg/L might have caused the problem.

6/22/2018
We found out that if we are testing fluoride solution with tap water, standard solutions for calibration should be done with tap water instead of DI water. In addition, we think that the insufficient calibration points might have caused the inaccuracy in the semilog model. Therefore, we should make 0.1, 0.5, 1, 5, 10, 15, 20mg/L of F- solutions for calibration.

[image]

cheertsang, 2018-06-22:
Calibration curve found [here](https://docs.google.com/spreadsheets/d/1Qdzn8rtu0ubeyHFeoxpHNJPTBkcbFvO21DLpOUB-Flc/edit?usp=sharing).

ChingPangggg, 2018-06-28:
Made fluoride calibration solutions of concentrations:
0.1 mg/L, 0.5 mg/L, 1 mg/L, 5 mg/L, 10 mg/L, 15 mg/L, 20 mg/L
(using tap water and TISAB)

closed 2018-08-10 by ChingPangggg