AguaClara/Fluoride-Auto #47: Run experiment with fluoride

cheertsang, 2018-06-25:
Started running system at 12:30 PM. 
Water pump: 51 rpm (0.76 mL/s)

The PACl flow rate is: 0.001583 milliliter / second
The PACl concentration in the stock is: 2400 milligram / liter
The PACl superstock volume in the stock is: 17.08 milliliter
The water volume in the PACl stock is : 482.9 milliliter

The fluoride flow rate is: 0.003167 milliliter / second
The fluoride concentration in the stock is: 720.1 milligram / liter
The fluoride volume in the stock is: 360 milliliter
The water volume in the stock is : 140 milliliter
The percent flow rate of fluoride/total flow through system: 0.4166 dimensionless %.
The total flowrate through the system is: 0.0007648 liter / second
The actual water pump flow rate required is: 0.7552 milliliter / second

ChingPangggg, 2018-06-25:
Checked at 2:30 pm (after 2 hours of experiment)
No floc formation is seen in the sedtube.

ChingPangggg, 2018-06-25:
Checked at 3:30 pm (after 3 hours)
Small floc formation is seen but no floc blanket is observed.

ChingPangggg, 2018-06-29:
6/29/2018
Water pump: 51 rpm (0.76 mL/s)
The PACl flow rate is: 0.001583 milliliter / second
The PACl concentration in the stock is: 2400 milligram / liter
The PACl superstock volume in the stock is: 17.08 milliliter
**The PACl concentration in the system is: 5 mg/L.**

The fluoride flow rate is: 0.00095 milliliter / second
The fluoride concentration in the stock is: 2400 milligram / liter
The fluoride volume in the stock is: 120 milliliter
The water volume in the stock is : 380 milliliter
**The fluoride concentration in the system: 3 mg/L**

ChingPangggg, 2018-06-29:
6/29/2018

Changed PACl dosage to 10 mg/L
The PACl flow rate is: 0.003167 milliliter / second

ChingPangggg, 2018-06-30:
6/30/2018

PACl Dosage: 15 mg/L
Fluoride: 3 mg/L

Gel was observed at the 5th hour of experiment
[Gel-like in sed-tube](https://raw.githubusercontent.com/AguaClara/fluoride/master/Summer%202018%20fluoride%20report/20180630_173850.jpg)

closed 2018-08-10 by ChingPangggg