AguaClara/Fluoride-Auto #51: 6/29 Fluoride 3 mg/L

cheertsang, 2018-07-01:
**PACl 10 mg/L**

Ran experiment for 3.5 hours.
Took 2 samples. 
No observed flocs in sedimentation tube.
Data and calibration curve found [here](https://docs.google.com/spreadsheets/d/1Qdzn8rtu0ubeyHFeoxpHNJPTBkcbFvO21DLpOUB-Flc/edit#gid=974397563).

cheertsang, 2018-07-01:
**PACl 15 mg/L**

Ran experiment for 6 hours. 
Took 5 samples. 
Flocs congealed into a gel in the sedimentation tube. Upflow velocity may be too slow. 
Data and calibration curve found [here](https://docs.google.com/spreadsheets/d/1Qdzn8rtu0ubeyHFeoxpHNJPTBkcbFvO21DLpOUB-Flc/edit#gid=974397563).

cheertsang, 2018-07-01:
**PACl 20 mg/L**
[image]

Ran experiment for 6 hours.

ChingPangggg, 2018-07-03:
**PACl 30 mg/L**
7/2/2018

ChingPangggg, 2018-07-03:
**PACl 40 mg/L**
7/3/2018

cheertsang, 2018-07-06:
PACl started to make a difference at 20 mg/L for 3 mg/L influent fluoride.

closed 2018-08-10 by ChingPangggg