AguaClara/Fluoride-Auto #54: Things to work on

cheertsang, 2018-07-13:
**Bench System**
- Determine the PACl concentration needed to form a proper floc blanket 
- find out what a proper floc blanket should look like

- Test error of fluoride probe
- Test concentrations of fluoride standards to see accuracy of fluoride probe
- Test 2-3 trials of each reading
- figure out how long to wait for each reading - graph concentration over time, see how long it takes to stabilize

- calculate residence time of sedimentation tube and flocculator to figure out how long it takes to reach desired influent concentration of fluoride

**Gravity-Powered System**
- add sliding adjustors to adjust effluent tube of the sedimentation tube instead of changing the height of the constant head tanks
- add platform for PACl and balance - figure out flow rate of coagulant

- test the different types of TISAB we have to see if it makes a difference

cheertsang, 2018-07-16:
[Residence time calculation](https://github.com/AguaClara/fluoride/blob/master/residence_time.md)
13.22 min

cheertsang, 2018-07-19:
5 rpm = The PACl flow rate is: 0.001583 milliliter / second
10 rpm = The PACl flow rate is: 0.003167 milliliter / second
20 rpm = The PACl flow rate is: 0.006333 milliliter / second
30 rpm = The PACl flow rate is: 0.0095 milliliter / second
40 rpm = The PACl flow rate is: 0.01267 milliliter / second
50 rpm = The PACl flow rate is: 0.01583 milliliter / second

cheertsang, 2018-08-01:
The PACl flow rate is: 0.0095 milliliter / second
The PACl concentration in the stock is: 2400 milligram / liter
The PACl superstock volume in the stock is: 17.08 milliliter
The water volume in the PACl stock is : 482.9 milliliter

The fluoride flow rate is: 0.00095 milliliter / second
The fluoride concentration in the stock is: 2400 milligram / liter
The fluoride volume in the stock is: 120 milliliter
The water volume in the stock is : 380 milliliter
The percent flow rate of fluoride/total flow through system: 0.125 dimensionless %.
The total flowrate through the system is: 0.0007705 liter / second
The actual water pump flow rate required is: 0.7496 milliliter / second

KevinSarmiento, 2018-08-06:
Final Tasks:

- [ ]

closed 2018-09-09 by cheertsang