AguaClara/Fluoride-Auto #56: Influent ad Effluent measurement

ChingPangggg, 2018-07-18:
Air bubbles will appear in the system (in the reactor) since we have to disconnect the tubing to take the influent sample. We think that this will have an effect in formation of floc blanket in the reactor. Therefore, we are thinking about  installing a t-connection and a valve after the fluoride is added and before the PACl enters the system.

cheertsang, 2018-07-19:
5 rpm = The PACl flow rate is: 0.001583 milliliter / second
10 rpm = The PACl flow rate is: 0.003167 milliliter / second
20 rpm = The PACl flow rate is: 0.006333 milliliter / second
30 rpm = The PACl flow rate is: 0.0095 milliliter / second
40 rpm = The PACl flow rate is: 0.01267 milliliter / second

ChingPangggg, 2018-07-27:
Valve was added after adding fluoride and before adding coagulant in the system.
[image]

closed 2018-08-10 by ChingPangggg