AguaClara/Fluoride-Auto #61: Meeting with summer team

tm545, 2018-09-06:
Hi,
Could we meet with you guys to catch up with work you've done over the summer and specifics about the probe and how much cooperation we should have between teams for the fall.

cheertsang, 2018-09-06:
Hi Tigran! Yes, that would be great! Let's all talk after Monday night class this week to set up a time?

cheertsang, 2018-09-08:
For now, [here](https://github.com/AguaClara/fluoride/issues/60) is where we left off in the summer. 

We ran tests with 5, 10, and 15 mg/L fluoride, while testing PACl amounts of 10, 20, 30, 40, 50. [Here](https://raw.githubusercontent.com/AguaClara/fluoride/master/Summer%202018%20fluoride%20report/adsorption_model.png) is a graph of our results. 
I would suggest running the same tests for 3 and 20 mg/L fluoride. 
@tm545 @SambaSowe @janakshah373