AguaClara/Fluoride-Auto #64: PACl adsorption mechanism

tm545, 2018-09-12:
Hi,
We're working on a langmuir isotherm for our adsorption model for fluoride. We just wanted to ask you about the general mechanism for how PACl binds to/adsorbs other molecules and if you could tell us about this process so we derive the correct isotherm to reflect this.
Thanks,
Fluoride Auto

monroews, 2018-09-12:
I don't know the mechanism. I do recall that the isotherm that the team created during the summer appeared to have a data analysis error in it. I suggest checking to make sure that the data points that are plotted on the isotherm were each calculated correctly.

cheertsang, 2018-09-12:
Just to be clear, here is how we plotted the data points on the isotherm:

**On the x-axis:** we plotted effluent fluoride (mg/L)
**On the y-axis:** we plotted (initial F - final F)/mass of PACl (mg/L)

[image]

cheertsang, 2018-09-12:
For your reference, [here](https://github.com/AguaClara/Fluoride-Auto/files/2376890/summer.fluoride.analysis.xlsx) is the Excel sheet we used to create this graph.

cheertsang, 2018-09-12:
Equilibrium partitioning between species in solution (Ce) and sorbed species (qe) is called a sorption isotherm or “isotherm”.

WPennock, 2018-09-12:
I have done a little reading on the subject, and it appears that adsorption with Al(OH)<sub>3</sub> is only one of the mechanisms, and the predominant mechanism (as of [2012](https://www.sciencedirect.com/science/article/pii/S0927775711007564)) appears to not yet be understood :

> In coagulation Al(OH)3 flocs forms during the hydrolization of Al salt and fluoride is removed by attachment to the flocs. In this process aluminum fluoride complexation (AlF complexation), ion exchange, electronic attraction and physical adsorption are involved [4]. But there is not a clear conclusion about the main mechanism.

A [later paper](https://www.sciencedirect.com/science/article/pii/S138358661630017X) by these authors goes into a more detailed exploration of the relative importance of the mechanisms at different values of pH. This might be a good place to start, since using PACl will not only result in adsorption, but also in complex formation.

However, to return to the original question, [this paper](https://www.sciencedirect.com/science/article/pii/S1385894711011569) expresses the (experimentally-informed) belief that adsorption of F onto AlOH<sub>3</sub> occurs by replacement of surface hydroxyl groups and gives adsorption characteristics for AlOH<sub>3</sub>:

> FTIR indicates the replacement of surface hydroxyl groups by fluoride.

I haven't read these papers in depth, but I'm hoping you can critically analyze them and draw conclusions accordingly. If there's any way I can help, please let me know.