AguaClara/Fluoride-Auto #67: Meeting with Sensor Dev team

janakshah373, 2018-09-24:
Hi Sensor Dev. We have an issue with our waste removal system for our current apparatus. Although we know the total flow rate in the waste stream, we are not capable of knowing the concentration or purity. Our goal is to know the amount of flocs that are part of the waste slurry being pumped out. This can be done using a guess and check method, but we would like to consider a more elegant, continuous solution. This would be to have a sensor based on Beer's law to turn a pump on or off based on the absorption through a floc weir. We would have to calibrate the probe to not turn on when a regular floc goes down the weir but will turn on the pump if flocs built up all the way up the weir.
If this is something you guys could give us some advice on, could we set up a meeting with your team soon?

lois-lee, 2018-09-24:
Why don't we set up a meeting so we can understand the premise of the problem better and we can go from there.

janakshah373, 2018-09-24:
What times are you free on Wednesday? We meet from 12-1 and 5-8.

lois-lee, 2018-09-24:
We don't generally meet on Wed, but I can talk to my group and see if they're free. If not, our team usually meets twice on Thurs

tm545, 2018-09-24:
Would Thursday at 4:15, Friday close to 1 be okay.

janakshah373, 2018-09-25:
That works, see you then!

closed 2018-09-25 by janakshah373