AguaClara/Fluoride-Auto #69: No Floc Blanket reactor

tm545, 2018-10-10:
Hi Will,
Monroe wanted us to see if our floc blanket effects our effluent fluoride concentration, so we were going to fabricate a new reactor that won't create a floc blanket similar to yours. Do you think we could ask you some questions about your design at some point.
Thanks,
Fluoride Auto

WPennock, 2018-10-13:
Hi, Tigran;

I am really interested in your project and would be glad to help in any way I can. I think I'm booked next Tuesday, but I should have availability on other days of the week. Let me know, and I hope I can help.

janakshah373, 2018-10-17:
Hi Will, we are here from 5-7 today, tomorrow from 4-6, and Friday from 12:30-1:30. Do any of those times work for you? If not, we could try to schedule an alternate time.

closed 2019-07-24 by tm545