AguaClara/Fluoride-Auto #80: Fluoride Probe Help

Melouie, 2019-06-03:
We are having trouble with the probe (brand new). We first made standard solutions and tried putting it in a 10 mg/L standard but it did not fluctuate or respond to shaking/stirring. We also found an online manual and made a 50mL TISAB and 50 mL DI water solution, then put 1 mL and 10 mL in there as per the instructions, but did not see a drastic change, which there should have been. We also changed the wire, the box, and closed out of ProCoDa. The R^2 value was calculated once (91%) but the rest of the times the voltages for the different concentrations were not different enough. We were wondering if you would be able to get the voltmeter box from the teaching lab for us to try? Do you have any other suggestions? (Tigran said that you suggested an acid solution a few semesters ago)

Thanks a lot!

monroews, 2019-06-04:
The first question is whether the blue signal conditioning box is working. Did you try both of those boxes that are in the AguaClara lab? 
We could confirm that the blue signal conditioning box works by trying it with a pH probe. If it works with the pH probe, then the problem is the Cole Parmer probe.
Did we ever have a Cole parmer probe that worked? If not, then, we should purchase a probe from the company that sold us the previous one that worked.

In the meantime, you can run experiments to work on the tasks we identified yesterday. A fluoride probe isn't necessary for all of those experiments.

monroews, 2019-06-06:
You can ask Paul if he can get you another blue signal conditioning box to test. If that isn't possible, then I suggest purchasing a new fluoride probe.