AguaClara/Fluoride-Auto #95: Capture Velocity Experimental Design

Melouie, 2019-11-17:
Hi Monroe, we were wondering if you would be able to check over our experimental design? We are trying to model effluent turbidity vs. capture velocity of flocs. Our current set up has a water tube going from the tap to the flocculator, and it gets dosed with fluoride and PACl before the flocculator. After the flocculator we have a t-junction, one end is an open valve that is our "effluent" and the other runs through a second pump, which models the capture velocity of the floc weir. We are running that pumped waste tubing through the turbidimeter (for the past few weeks we have been trying to model turbidity vs. influent PACl concentration, so we are hoping this experiment will match up with those numbers). After the turbidimeter, we measure fluoride concentration. The point of the experiment is to determine the lowest capture velocity possible in order to remove the most flocs. If there are any details I missed please let me know! Thank you!

monroews, 2019-11-17:
Is there a tube settler before the turbidimeter?

Melouie, 2019-11-18:
No, we are using the tubing through the second pump as the tube settler.

monroews, 2019-11-18:
The tube settler is the sloped section of the sedimentation tank.

Melouie, 2019-11-18:
We don't have a sedimentation tank in the setup (we were trying to model the big horizontal sedimentation tank that was in the grad lab, but we used tubing instead of PVC since we got gelling in the summer when we tried it with the PVC).

monroews, 2019-11-19:
I'm not sure about this and I don't remember how this design was supposed to work. I don't recall a horizontal flow sedimentation tank. The sloped tube settler is made by installing a T in a horizontal pipe coming from the flocculator. The "T" is placed upside down. The  leg of the "T" is angled at 60 degrees.

Melouie, 2019-11-20:
I think when we designed the experiment we were trying to make it completely independent of angle (since the angle was part of the sedimentation tube design and we wanted to redesign it, so we thought the capture velocity would be determined by the pump regardless of angle), but none of us are that comfortable with capture velocity. We talked about it with Justin during our technical discussion on Monday and we might change the experimental design when we come to lab this week. We will keep you posted!